cwi-dis/iotsa #27: ESP01 https mode

mehotkhan, 2020-08-15:
hi , thanks for your work
i try to install the framework on ESP8266 (01)
Hello API work but when I enable HTTPS mode, give me build error
```
Linking .pio/build/esp01-Api/firmware.elf
/home/ali/.platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: .pio/build/esp01-Api/firmware.elf section `.irom0.text' will not fit in region `irom0_0_seg'
collect2: error: ld returned 1 exit status
*** [.pio/build/esp01-Api/firmware.elf] Error 1
```

mehotkhan, 2020-08-15:
ok, i missed some platformio confirg, and https work perfectly on esp01 
```
[esp8266]
extends = common
platform = espressif8266
board = esp01_1m

[https-api]
src_filter = +<*> +<../examples/httpsApi>
build_flags = -DVTABLES_IN_FLASH -DIOTSA_WITH_HTTPS -DIOTSA_WITH_COAP
```

jackjansen, 2020-08-16:
Good to hear you got it working! I'll close this issue then, please reopen if you disagree.

closed 2020-08-16 by jackjansen