cwi-dis/iotsa #28: implement WSS

mehotkhan, 2020-08-19:
it's possible to implement `WSS` IOTSA framework?
 with https cert/key?
+ I need data to communicate with ESP and browser, on mDns domain
+ COAP is a good choice but it seems not possible to use UDP in the browser ( there are no browser js clients )

jackjansen, 2020-08-19:
Let me ask you for help:-)

I have (for a long time) been thinking about adding streaming interfaces (wss/websockets is an option, but also http push events, or raw TCP sockets) to iotsa, but I can't think of a way to design it in an elegant way.

The problem is that the whole design of iotsa is state-based, with the client getting or setting variables, so it fits the REST (or COAP) design very nicely. Callbacks (from iotsa to something else on the net) are somewhat handled by the `IotsaRequest` module, and I know how I want to do this better (when I have the time:-)

But streaming data from client to iotsa or from iotsa to another agent on the net is something that I don't know how to fit in the architecture, in a decent way.

How would you want to use WSS? With iotsa in control (for example transmitting new sensor readings whenever available) or with the other side in control? Or both? How would you envision integrating it in the existing architecture?

mehotkhan, 2020-08-20:
my User App write in js and load with a browser (PWA app + Https + WASM)
for now , I use WS to send sensor `data -> ESP01 -> WIFI -> Browser -> PWA app`

for security reason, `WS` blocked on https domain, 
my Dev code uploaded here :

[ESP01-Signaling-Server](https://github.com/konar-studio/ESP01-Signaling-Server/blob/master/esp01-ws-ap.ino)
`JSON data read from serial, send to WS client`
+ my  main purpose:  data must stream from sensor to js app, as fast as possible 
+ data must be encrypted in some way (based on the protocol )
+ I tested MQTT before on cloud service, MQTT need a broker, I find this, not tested : [uMQTTBroker](https://github.com/martin-ger/uMQTTBroker)
+ COAP, no UDP on the browser, find this code `https://github.com/jdeblese/jscoap/wiki/API`,7 years old and don't work
I'm working on this js module, maybe in some way, I make it work 

+ your Framework has a lot potential for ESP01 and modularity philosophy, I think on COAP TCP mode (i think your COAP API in TCP 
 mode), maybe I achieve that
+ my final idea is , in some way create  WSS with your Framework or find a machine to machine protocol that works on Arduino + JS Browser
+my main language is not English, Grammarly help me, sorry for confusing text :))

jackjansen, 2021-01-01:
@mehotkhan did you make any progress on WSS? As said: I'd love to incorporate the functionality in iotsa, and I'm working on a new release at the moment, so if you have anything you'd be willing to share: right now I could use sample code (even if incomplete) if you're willing to share...

mehotkhan, 2021-01-02:
hi, jack,
I change the main plan to : 
create Secure custom Event source,
Every path in un event source, so the webserver can send data on-alive
read here :[async-event-source-plugin]( https://github.com/me-no-dev/ESPAsyncWebServer#async-event-source-plugin)

the main concept for using this method is: update server code something like 

```
<?php
header('Content-Type: text/event-stream');
header('Cache-Control: no-cache');

$time = date('r');
echo "data: The server time is: {$time}\n\n";
flush();
?>
```
I read and about some code but I can't find a way to do this on iotsa
and I lost my free time :))