cwi-dis/iotsa #30: Build errors on ESP-12

reneklootwijk, 2020-11-11:
Building iotsa v2.4.1 for the ESP-12F on PlatformIO results in the following fatal errors:

In file included from .pio/libdeps/nodemcu/esp32_https_server/src/ConnectionContext.cpp:1:0:
.pio/libdeps/nodemcu/esp32_https_server/src/ConnectionContext.hpp:8:25: fatal error: openssl/ssl.h: No such file or directory

.pio/libdeps/nodemcu/esp32_https_server/src/HTTPConnection.hpp:8:28: fatal error: mbedtls/base64.h: No such file or directory

Does anybody know how to resolve this?

jackjansen, 2020-11-12:
Hmm.... You're building for esp8266, but somehow esp32_https_server is included in the dependencies (which it shouldn't be, because it is esp32-only).

Could you try using the head of the _develop_ branch, because I vaguely remember I've seen this issue and fixed it...

If that does not fix it: please provide the exact version of platformio and the exact version of the esp8266 arduino framework you're using, then I can try to repeat the problem.

mehotkhan, 2020-11-12:
test this platformio.ini code , for esp01 , and write your module code on app folder 
```

[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino
monitor_speed = 115200
board_build.f_cpu = 160000000L
lib_ldf_mode = deep+
lib_deps = 
    Time@~1.6
    Timezone@~1.2.4
    ArduinoJson@~6.15.1
    ArduinoJWT@~1.0.1
src_filter = +<*> +<../app>
build_flags = 
    -DVTABLES_IN_FLASH
    -Wdeprecated-declarations
    -DIOTSA_WITH_HTTPS

[use_debug_prints]
build_flags =  -DCORE_DEBUG_LEVEL=6

[use_debugger]
debug_tool = minimodule
upload_protocol = minimodule 

```

reneklootwijk, 2020-11-12:
> Hmm.... You're building for esp8266, but somehow esp32_https_server is included in the dependencies (which it shouldn't be, because it is esp32-only).
>
> Could you try using the head of the _develop_ [...]

This did not fix the issue. I am using version 5.0.2 of PlatformIO. How can I determine which version of the Arduino framework is being used?

reneklootwijk, 2020-11-12:
> test this platformio.ini code , for esp12e(esp 8266) , and write your module code on app folder
> 
> ```
> 
> [env:esp12e]
> platform = espressif8266
> board = esp12e
> framework = arduino
> monitor_speed = 115200
> board_build.f_cpu = 160000000L
> lib_ldf_mode = deep+esp12e
> lib_deps = 
>     Time@~1.6
>     Timezone@~1.2.4
>     ArduinoJson@~6.15.1
>     ArduinoJWT@~1.0.1
> src_filter = +<*> +<../app>
> build_flags = 
>     -DVTABLES_IN_FLASH
>     -Wdeprecated-declarations
>     -DIOTSA_WITH_HTTPS
> 
> [use_debug_prints]
> build_flags =  -DCORE_DEBUG_LEVEL=6
> 
> [use_debugger]
> debug_tool = minimodule
> upload_protocol = minimodule 
> ```
> 
> [image]

This worked although I had to change ib_ldf_mode = deep+esp12e to ib_ldf_mode = deep

Thanks!

jackjansen, 2020-11-16:
The `lib_ldf_mode = deep` is indeed what should fix it. Thanks @mehotkhan for finding it, I had completely forgotten (because I always use mode `deep` or `deep+` myself, after having been bitten)

The underlying problem is with the PlatformIO LDF (Library Dependency Finder). It analyses toplevel `.h` files (or `.h` and `.cpp` files if you specify the `+` in the mode) to find packages on which iotsa depends. But this accidentally finds the esp32_https_server even if you're building for esp8266 (where it isn't needed and doesn't exist).

Assigning myself so I remember to put a strong note in the documentation about this. (but if you have suggestions on how this could be fixed better: please let me know).

reneklootwijk, 2020-11-20:
Whatever I try, I keep getting the same error.

reneklootwijk, 2020-11-20:
Finally got it working. The first time I build a dependencies will be installed in the .pio/libdeps/<board> directory, including the  esp32_https_server and esp32_https_server_compat libraries. When these libraries are deleted manually, the build is succesful.

When all directories in the .pio/libdeps/<board> directory are deleted, the build will import all dependencies again including the esp32_http directories resulting in a failing build, despite the lib_ldf_mode = deep+ option.

reneklootwijk, 2020-11-20:
Adding lib_ignore = esp32_https_server_compat to platformio.ini prevents the esp32_https_* libraries from installing. It is a workaround of what seems a bug in platformio, but it works.

jackjansen, 2021-01-02:
Basically need to document this: I don't think there's a workaround around the platformio underlying issue that somehow `lib_ldf_mode = deep+` isn't smart enough to see that the whole `esp32_https_server` isn't used:-(

jackjansen, 2022-08-01:
Finally fixed this in 22d24d2bac. It turns out there is a config variable `lib_compat_mode` that you need to set to `strict` if you want the library finder to keep track of the architecture you are building for. If you don't specify this value the default is to only check whether the library is compatible with the platform used.

The `library.json` for iotsa now states correctly which dependency library is needed for which architecture, so specifying 

```
lib_compat_mode = strict
lib_ldf_mode = deep+
```

should finally do the right thing.

closed 2022-08-01 by jackjansen